thiagoralves/OpenPLC_v3 #1: Modbus Configuration File Location

zbirnba1, 2018-07-25:
Using openplc_v3 how do you upload a modbus configuration file? I tried following the directions found [here](http://www.openplcproject.com/getting-started-modbus-io) but didn't see a place in the GUI to upload the new config (as there was in openplc v2).

I followed the install directions for v3, but unlike in v2, there was no option to activate the modbus driver.

thiagoralves, 2018-07-25:
On v3 the modbus configuration will be done directly on the web interface on the Slave Devices menu. It won't need a configuration file anymore. However this functionality is not ready yet, so for now, to use modbus you still need to stick with v2.

thiagoralves, 2018-08-15:
> Just tried to use the new functionality, webproxy gave the error "The
> browser (or proxy) sent a request that this server could not understand.".

Can you tell me exactly the steps you took to reproduce this problem?

zbirnba1, 2018-08-16:
To install:

```
git clone https://github.com/thiagoralves/OpenPLC_v3.git
cd OpenPLC_v3
./install.sh linux
./start_openplc.sh
```
I first logged in (no issues), then tried to add a new modbus device by going to the "slave devices" menu.  I did this prior to uploading a program to the vPLC.  However I tested with a basic program uploaded to the PLC and got the same results.

I entered the following information into the menu:
Device Name: Device 0
Device Type: Generic Modbus TCP Device
Slave ID: 0
IP Address: #########
IP Port: 502
Discrete Inputs Start: 0
Discrete Inputs Size: 8
Coils Start: 0
Coils Size: 8
Input Registers Start: 0
Input Registers Size: 8
Holding Registers-Read Start: 0
Holding Registers-Read Size: 8
Holding Registers-Write Start: 0
Holding Registers-Write Size: 8

When I selected "save device", it then pointed me to the error message "The browser (or proxy) sent a request that this server could not understand".

Is there any logging I can enable to provide more information?

thiagoralves, 2018-08-17:
I found the problem and fixed it today. Can you please clone the code again, reinstall and test? Thanks.

closed 2018-08-20 by thiagoralves