thiagoralves/OpenPLC_v3 #2: V3 Eating CPU

LunarMongoose, 2018-08-08:
I am having this issue where the third version of OpenPLC maxes out two cores on my machine. I have run it in many different environments. An 8 core xeon processor, a two core VM, and an i7 machine all of them, while running Openplc have two cores maxed out. To me it seems like this shouldn't happen? Have you seen the same results? Do you have a solution? Installed using the linux option on the install script and running the blank program

pedromorgan, 2018-08-17:
Similar here, for four cores and two were fluttering around 100%

thiagoralves, 2018-08-17:
OpenPLC v3 starts many more threads than v2. Maybe one or two of those
threads are not behaving well and eating a lot of CPU. I will take a look
at this.

thiagoralves, 2018-08-17:
I just found the problem and fixed it. It was related to the modbus and interactive server threads. They were running on an infinite loop without a sleep. I just added a sleep in between and now v3 uses around 37% CPU on a Raspberry Pi with DNP3 enabled. If you disable DNP3, CPU usage drops to something around 1%

closed 2018-08-17 by thiagoralves