thiagoralves/OpenPLC_v3 #5: database + schema

pedromorgan, 2018-08-18:
So I realise that the 
- sqlite db is in version control 
-  and cannot see any schema creation scripts ( do they exists )

Guess the db needs to be created on setup and places in users home dir etc

pedromorgan, 2018-08-18:
Also suggest using dict for row's returned instead of tuple, as its less confusing, imho
ie `row['foo']` instead of `row[4]`
https://stackoverflow.com/questions/3300464/how-can-i-get-dict-from-sqlite-query

pedromorgan, 2018-08-18:
Ok so after so thought, and thinking out loud,

do we really needs sqlite ??

can't this stuff be stored in a "workspace with folders" in json, yaml, toml etc ??
indeed a per project folder etc..

Whats limiting about sqlite ??
maintaining future different db's different schema versions etc...
eg query crash cos table not htere...
Whereas if it was in nosql = files (in git even)
then the whole lot can be handled better..

pedromorgan, 2018-08-18:
Also check http://json-schema.org/

Is kinda maube an idea..
or whatever ptorocl files can grap from internet.

thiagoralves, 2018-08-20:
Initially I didn't want to use a database. I was going to store everything on CSV files (I know, hardware guys love CSV). But then I though a db would make my life a lot easier, so I went with the simplest one available. However, JSON might be more interesting. The only problem is to translate the whole project to use JSON instead of db.

thiagoralves, 2019-02-01:
I'll close this for now until there is more progress on the subject...

closed 2019-02-01 by thiagoralves