thiagoralves/OpenPLC_v3 #6: golang

pedromorgan, 2018-08-18:
Ok so this might be crazy..

But the more I dig into the code,
I realise there is a lot of comms stuff.. and alike
going on etc etc..

This smacks of what golang is good at, concurrency, multi things going on, websockets et all..

Would be intrigued to start a project as altername, for fun of course

pedromorgan, 2018-08-18:
reasons..

its static types, so no cock up on stuff..

pedromorgan, 2018-08-18:
Be interesting ultimately to create a single page app..
with websockets and background ajax..

Am also developing a NativeScript app for mobile, so kinda r+d there also

pedromorgan, 2018-08-24:
Ok so working on golang version, for my curious...

thiagoralves, 2019-02-01:
I'll close this for now until there is more progress on the subject...

closed 2019-02-01 by thiagoralves