thiagoralves/OpenPLC_v3 #8: Manual + Documentation

pedromorgan, 2018-08-18:
Docs =  generate api and stuff from source code..

Its a complex project with c++, py, sh et all..
To skin the cos cat,
I suggest firing doxygen at the source..

Doxygen is cool cos it parses cpp + c initially,
but also md and many others...

And automate it..

This would give the api docs..

pedromorgan, 2018-08-18:
So jumped the gun and registerd 
https://github.com/openplcproject

as a project..

thiagoralves, 2018-08-20:
Yes, website is wix (sorry, poor www skills forced me to do that! =} ). We can create a wiki page for documentation and link it to the wix website. So people who want to know more about the project will be redirected to the wiki page. Noob users who just want to use the thing can just stay in the wix part and download/install it.

pedromorgan, 2018-08-24:
Ok so creating the docs repos...

Main snag is that we want the docs to be current.. but also legacy..

that is what https://github.com/openplcproject/meta
which will contain content from
- http://www.openplcproject.com/
- and from source code..

and publish at openplcproject.github.io = TLD in setting..

Main thing, is we need to
1) tie the docs to the the code
2) different versions etc ?? is that likely to change in much ?

pedromorgan, 2018-10-17:
So i just setup a couple of sphinx projects on gitlab..
and it all works..
basically a git push, recompiles all the does etc..
eg
https://open-geotechnical.gitlab.io/aggregator/
and the "builds" at
https://gitlab.com/open-geotechnical/aggregator/-/jobs

So this is what we need..
and with the c++ etc..
we can knock it oout with doxygen
https://gitlab.com/pages/doxygen

Only snag it that github.com does not do this stuff ;-(

pedromorgan, 2018-10-17:
So here's proof of concept..

https://openplc.gitlab.io/openplc_v3/md_README.html

If u like this idea, i will pursue..
means migrating to gitlab though,,..

indeed the #14 CI experiment also

thiagoralves, 2019-02-01:
I like the idea of documenting in directly in the code. Let's see what we can do on the future. I'll close this for now until there is more progress on the subject...

closed 2019-02-01 by thiagoralves