thiagoralves/OpenPLC_v3 #10: naming.. and namespace

pedromorgan, 2018-08-19:
Just to hit on..

Namespace,

so in url's an using
/programs
/program/<id>/start|stop

We need to namspace the stuff..

so maybe 
`oplc` is good for me..

pedromorgan, 2018-08-19:
What we want are slaves and a factory namespace...

eg
/wales/teilo_factory/knitting/needles
might return the no of needs in stock part no etc..
and that what I mean with ns.. kinda.. but a webdev, so mainly terminology...

eg
/factory/{line}/status

so we got 3 lines
- /factory/1/status
- /factory/2/status
- /factory/3/status

Thats we all what they all want init..

pedromorgan, 2018-08-24:
Explain again..

On the www interface we have a loaded projects..
and start run..

So if we have 3 projects running, they need their own "address" and namespace not ?

its an url issue..
server:/

closed 2018-08-24 by pedromorgan

thiagoralves, 2018-08-25:
>> So if we have 3 projects running...
On OpenPLC v3 you can upload multiple projects, but you can only run one project at a time. It makes no sense to run multiple projects together. A project is a whole PLC instance, and that’s it.