thiagoralves/OpenPLC_v3 #9: Split projects

pedromorgan, 2018-08-19:
Ok so the c++ and stuff makes me nervous..

But the project should be spit up anyway..

- the runtime
- the monitor..

pedromorgan, 2018-08-19:
What I mean is that atmo..
- the whole caper is in one repos. monolithic
- whilst the python/flask frontend is in python intermixed with cpp+machine code etc

So really there are a few bits to get up and running for future proof..

- split the PLC c++ stuff into a seperate repos eg `openplc3-cpp`
-  extract the "webserver bits" into a repos eg `oplc-www-py`
-  @pedromorgan is already thinking about  golang eg `oplc-go`

Also need the docs project, to make it simple for muppets, like me ;-)

pedromorgan, 2018-08-19:
Snag is @thiagoralves been here before.. so maybe i confuse you.. and apologise .. sorry..

But ..
- u keep hacking on core stuff down there
- and us not there harware style.. we overview on www interface init..

thiagoralves, 2018-08-20:
I see your point. However, since the www stuff deals directly with bash scripts and the c++ code, I don't think it would be a good move to split them up. Plus, it will be one more repository to clone for people that want to install OpenPLC. I tried to isolate them in different folders, so that you don't need to mess around with the machine code to work on the frontend. But I'm open for discussion, I'm not saying my organization is the best, it just works well for me I think.

thiagoralves, 2019-02-01:
I'll close this for now. I believe the project should stay combined.

closed 2019-02-01 by thiagoralves