thiagoralves/OpenPLC_v3 #14: CI experiment

pedromorgan, 2018-10-17:
I want to try an experiment,
of building openplc agains different libs version..

Can u give me a hint as to which are "externals"
eg

dnp3 is here https://github.com/automatak/dnp3 and all the "tags"

which are the other externals.. ?

libmodbus = https://github.com/stephane/libmodbus
matiec = https://bitbucket.org/mjsousa/matiec (NO TAGS)
?

thiagoralves, 2019-02-01:
Yes, those are the correct locations. There are all libs you will need.

closed 2019-02-01 by thiagoralves

shrmrf, 2019-02-05:
@pedromorgan @thiagoralves did you guys get the CI/CD thing running? 

I can help set up the builds if you want. Maybe we should also publish official docker images on Docker Hub?

pedromorgan, 2019-02-05:
@shrmrf I got a "basic" working on gitlab, but would be cool to have docker (altought never used as yet)

https://gitlab.com/openplcproject/openplc_v3/blob/master/.gitlab-ci.yml

pedromorgan, 2019-02-06:
@shrmrf @thiagoralves  gitlab allows for different builds jobs with images libs etc (example below)

This also means we can test against differrent version of eg dnp3, modbus etc

eg
```
before_script:
   - apt-get install --yes cmake libmatio-dev libblas-dev libsqlite3-dev libcurl4-openssl-dev
   - apt-get install --yes libarchive-dev liblzma-dev

build:precise:
  image: precise:base
  script:
    - mkdir build/
    - cd build
    - cmake -D CMAKE_BUILD_TYPE=Debug ../
    - make

build:trusty:
  image: trusty:base
  script:
    - mkdir build/
    - cd build
    - cmake -D CMAKE_BUILD_TYPE=Debug ../
    - make
```

shrmrf, 2019-02-06:
@pedromorgan yes, we can do the same thing with travis as well. It's what they call a build matrix. I think all CI tools are created equal :) whatever Thiago decides wins in the end

thiagoralves, 2019-02-06:
Well, I'm definitely not the CI expert in here, but I would like to use the
least amount of tools possible, just to avoid confusion. If all we need is
available in travis, lets just use this then. However, I like the
doxygen-like documentation that Pete generated on gitlab.

pedromorgan, 2019-02-06:
Thats the main reason for gitlab. does the same as travis + much more...

Its more akin to Jenkins, as builds docs, assemble tarballs et all

shrmrf, 2019-02-06:
I'm fine with Gitlab's CI too .... It's actually what we use at work. But if you enable `gh-pages` on Github, we can generate the Doxygen output from travis as well.

I personally just default to travis as it's more widely used in OSS. We're living in CI heaven these days: there are soooo many options (azure, circle, jenkins, gitlab, github's own yes they too have CI and many many more) but if you're looking for OSS support, travis is by far the most prevalent

thiagoralves, 2019-02-06:
I'm liking travis so far. The integration with github seems legit! If we
can generate Doxygen from travis as well, lets use it then as our main CI
for linux builds.

pedromorgan, 2019-02-06:
So here's gitlab attemping to build (aint got much time atmo)..

cygwin+linux+rpi
https://gitlab.com/openplcproject/openplc_v3/-/jobs

https://gitlab.com/openplcproject/openplc_v3/blob/ci_test/.gitlab-ci.yml

pedromorgan, 2019-02-06:
@thiagoralves u can use travis to build docs, but then its a total pain in the butt to setup travis, to make a tarball, set up keys, hooks , publish.. argh! etc... 

basically with GH, anything not static or jekyll (with NO extra plugins) is no no

shrmrf, 2019-02-06:
@thiagoralves https://shrmrf.github.io/OpenPLC_v3/ (doxygen generated docs) ... [changes for deploying on GH Pages](https://github.com/shrmrf/OpenPLC_v3/blob/master/.travis.yml#L15-L22)

@pedromorgan If you think Gitlab has a better CI story, I would vote for it! Let's go for Gitlab CI but let's not make a copy of this repo there, we can set up Gitlab CI/CD pipeline without the need to mirror this repo... i.e. we should still be pushing to _this_ repo on Github... (see the example of my CI/CD pipeline here: https://gitlab.com/shrmrf/OpenPLC_v3 )

Build badges from Gitlab can be added instead of the one we have from travis in README.md

@thiagoralves let's go for Gitlab.

shrmrf, 2019-02-06:
Go to Gitlab and click on the + button:
[image]

Click on **New Project**

Select the Tab: **Ci/CD for external Repo** and then click on Github and then it's not too complex afterwards

[image]

Let me know if I can help

pedromorgan, 2019-02-06:
The "group" is there already..
https://gitlab.com/openplcproject

shrmrf, 2019-02-06:
Yes, but I don't think it's synced/mirrored exactly... we can use the group of course, but the repo should just be a mirror over there and we shouldn't edit it there IMO.