thiagoralves/OpenPLC_v3 #17: Unclear license status

zagor, 2018-11-07:
The libraries in utils/ each contain their license declarations, but the core openplc python code in webserver/ doesn't mention any license.

moontide, 2018-12-28:
/me want to know the license information too, such as can I use OpenPLC (Runtime + Editor + HMI  builder) in commercial projects?

shrmrf, 2019-02-01:
Yeah, would be good to have a `LICENSE` file at the root. I'd lean towards GPLv3!

shrmrf, 2019-02-01:
@thiagoralves in Summary, if you're referencing other projects you don't need to take care of their license. From my cursory look at the sources, I see that you're linking to shared objects, so it might be a good idea to remove the sources from this repo and add a reference (`SHA-1` of the commit e.g. or submodule). This way you're fairly safe.

OTOH, I'm no lawyer.

shrmrf, 2019-02-01:
I believe that the using the GPL is fairly safe all the time because it's very copy left. It's also beneficial for you because it protects you from liability in case a nut-case of a company decides that OpenPLC was the reason of financial loss for them.

pedromorgan, 2019-04-26:
@thiagoralves 
All the loicence and stuff in core and stuff is a headache...

also they may conflict..

pedromorgan, 2019-07-25:
Ok so maybe closed soon with Apache2 license..
https://github.com/thiagoralves/OpenPLC_v3/pull/71

garretfick, 2019-07-31:
@zagor @moontide @shrmrf I've spent some time working on licensing (with Thiago's approval) and hopefully things in the development branch make more sense. Would you talk a look to see if the current license make sense?

shrmrf, 2019-07-31:
Hey @garretfick all licenses look good to me :) Having one or more (at least for my company) is better than not having any.

I took a cursory look and I don't think I can say much. But for any commercial company to be able to even touch this code, a `LICENSE` file of some sort is very important. Thanks for working on this bit!

garretfick, 2019-07-31:
I'm working on this from a similar perspective, making it very important for us as well.

garretfick, 2019-11-09:
@zagor Is this sill an issue for you, particularly in the development branch. We've been working to clarify licensing over year. I think there is still work to do, but would appreciate your feedback on where to focus.

closed 2020-05-02 by garretfick