thiagoralves/OpenPLC_v3 #22: Fix Python Dependency versions

shrmrf, 2019-02-01:
@thiagoralves what is your opinion on fixing the Python dependencies through a `requirements.txt` file?

This way we won't break if for instance the Flask project has some changes in the latest version. We can just issue a `pip install -r requirements.txt` command for installation of the dependencies.

I can submit a PR if you're interested.

pedromorgan, 2019-02-01:
@shrmrf yep..

pedromorgan, 2019-02-01:
Also we need to consider python3 (although need to check flask etc)

shrmrf, 2019-02-01:
@pedromorgan sure... Python 2 is going out in 2020 so not a bad idea :)

closed 2019-02-01 by thiagoralves

thiagoralves, 2019-02-01:
I think the requirements.txt file is a good idea. I saw that you made the modification only on the Docker install. Would you mind modifying it for all the other platforms as well? We might just need to make sure it will work under Cygwin (Windows) as well...

About python3, have you tried just running it plain in python3 to see what happens? I still haven't, but it may be worth trying

thiagoralves, 2019-02-01:
In fact, I'm starting to get a bit worried about this. Github notified me after I accepted the requirements.txt file that it is using pycrypto <= 2.6.1 and there are security issues about that. In that sense, having pip install downloading the latest version of packages might be a good idea. Maybe we should keep the requirements.txt file as a backup in case something breaks in the future...

shrmrf, 2019-02-01:
Another perspective:

>In fact, I'm starting to get a bit worried about this. Github notified me after I accepted the requirements.txt file that it is using pycrypto <= 2.6.1 ...

This I would say is an advantage of using `requirements.txt`.. GitHub can understand and parse this file. Otherwise, we'd never know of a vulnerability even though it was there. ([more on this from GitHub's blog](https://github.blog/2018-07-12-security-vulnerability-alerts-for-python/#how-to-start-using-python-security-alerts)

shrmrf, 2019-02-02:
FYI: I did a bit of investigation and figured that `pycrypto==2.6.1` is installed by just installing `python` and `python-pip` i.e. `sudo apt install python python-pip`

Thus, I think removing it in the PR #24 was the correct idea. 

Sorry again for the alarm I caused.