thiagoralves/OpenPLC_v3 #25: Mirror HMI Builder and Editor

shrmrf, 2019-02-04:
@thiagoralves would it be a good idea to put the source-code for the HMI builder and editor projects on GitHub?

thiagoralves, 2019-02-05:
The editor has its own repository on bitbucket: https://bitbucket.org/automforge/beremiz
The HMI has its code on SourceForge: https://sourceforge.net/projects/scadabr/

Since these two are mainly maintained by other groups, I don't want to just clone them here unless I have someone actively working on them for the benefit of the OpenPLC project.

closed 2019-02-05 by thiagoralves

shrmrf, 2019-02-05:
BTW, I created https://github.com/shrmrf/ScadaBR because I don't trust Sourceforge to survive for long with that 80's web interface :)

The latest release can be found here: https://github.com/shrmrf/ScadaBR/releases/tag/v0.1.0

pedromorgan, 2019-02-05:
@shrmrf we have similar thoughts...

I registered the openplcproject on GitHUB with a view to mirror/fork libs in one place
https://github.com/openplcproject

I then started to build the docs, and realised that gitHub has it limitations...(ie only jekyll)
So i set up a project at gitLAB.. 
https://gitlab.com/openplcproject
and the result is here
https://openplcproject.gitlab.io/openplc_v3/

Unfortunately, am busy on a contract and no time (or indeed spoken to @thiagoralves ),
but my goal is to have all the "projects" in one place, and run tests, build docs, docket etc all in one place and automated ;-)

shrmrf, 2019-02-06:
@pedromorgan coool! I just started playing with OpenPLC like 2 weeks ago and love the efforts @thiagoralves has put into the project. I'm looking for ways (after work hours) to help around.

Btw, the gitlab links are returning a 404

pedromorgan, 2019-02-06:
@thiagoralves u ok with using gitlab ? mainly cos it better Ci and can use doxygen.jekyll et all to create www pages etc

shrmrf, 2019-02-06:
I don't mind. I'm using it in my job at the moment.... although I also submitted a PR (#26 )... feel free to reject it

thiagoralves, 2019-02-06:
I don't mind using gitlab either, but does it mean that we'll have to move the entire repository to gitlab instead of github? I'm just thinking about changing the install instructions for every platform on the website to use the git clone command on a different address.... =/

shrmrf, 2019-02-06:
@thiagoralves no, we don't need to move the repo from Github for Gitlab CI. It's the same thing as Travis (though the setup involves managing a couple of secrets - no biggy).

OTOH, I like Travis' integration with Github 

I think we should look into appveyor to address the Windows builds.

thiagoralves, 2019-02-06:
The Windows build runs on top of Cygwin, so the test script needs to
install that on the VM before the test actually starts. Would that be
possible on appveyor? Again, I think we should try to focus our attention
in one single tool if possible. So let's examine all of them and see which
one would work best. About the Windows stuff, since it requires specific
configurations, I can test it myself manually after every commit, that's no
big deal. The Windows installer creates a shortcut to rebase OpenPLC
installation, which gets the most recent code from github and reinstalls
everything. That's all it takes to test it under Windows.

shrmrf, 2019-02-06:
@thiagoralves the common practice is to have `.travis.yml` and `appveyor.yml` in the repo as they're the most popular for Linux/Mac and Windows builds respectively. (As an example, [see the rust repo](https://github.com/rust-lang/rust).. it's one of the best managed projects out there*)

_\* a little caveat, they're actually going to soon move away from travis because there needs are by far excessive - not our problem_

shrmrf, 2019-02-06:
As I mentioned in #28 we can try this out if you give me the blessings, in my fork over the weekend. A cursory reading of the docs shows that it definitely _is_ possible

thiagoralves, 2019-02-06:
Great. Try appveyor in your fork. If that works well, let me know.