thiagoralves/OpenPLC_v3 #28: Windows CI Pipeline

shrmrf, 2019-02-06:
@pedromorgan and @thiagoralves 
As we discussed in #14 and #26 we should consider a CI pipeline on Windows as well. I know for a fact that appveyor is the most popular CI tool for Windows builds but I personally have no experience with it.

If it's fine with you guys, we can start an experiment in my fork to add appveyor support?

pedromorgan, 2019-02-06:
Well at least the rpi job succeded.. https://gitlab.com/openplcproject/openplc_v3/-/jobs/157437170

the cygwin needs tweaking i think

shrmrf, 2019-02-06:
@pedromorgan does Gitlab also have the option of running Windows builds? (that'd be awesome)

pedromorgan, 2019-02-06:
Its running a cygwin build https://gitlab.com/openplcproject/openplc_v3/-/jobs/157437169

shrmrf, 2019-02-06:
May I suggest using the official raspbian image for the rpi?: https://hub.docker.com/r/raspbian/jessie

pedromorgan, 2019-02-06:
@shrmrf umm https://gitlab.com/openplcproject/openplc_v3/-/jobs/157463721

pedromorgan, 2019-02-06:
@thiagoralves Whats your gitlab user ??

shrmrf, 2019-02-06:
>@shrmrf umm https://gitlab.com/openplcproject/openplc_v3/-/jobs/157463721

:confused:  I guess we should revert. Sorry about that :(

thiagoralves, 2019-02-06:
I don't have a gitlab user yet. Will create an account soon, just in the
middle of a big mess right now...

pedromorgan, 2019-02-06:
@shrmrf I'll look at it later..

Also setup a temp chat room here https://gitter.im/openPLCProjectx/community

shrmrf, 2019-02-06:
This container (`teaci/cygwin64`) isn't a Windows container: https://gitter.im/openPLCProjectx/community?at=5c5b44e68790df06206de3b2

I think appveyor is our last hope

pedromorgan, 2019-02-06:
It looks like appveyor in tergrated with gitlab.. just trying to figure out the yml file format..

pedromorgan, 2019-02-06:
somethings like this https://gitlab.com/inkscape/inkscape/blob/master/.appveyor.yml

closed 2019-02-08 by shrmrf