thiagoralves/OpenPLC_v3 #36: new www server

pedromorgan, 2019-03-26:
Hi @thiagoralves 

Got some time so hacking away at last part of www server merge.. #3 

Am also looking at dumping sqlite and using files instead (json etc)

You mention csv files for storage,
so for hacking purposes do u have a few samples or the format ??

pedromorgan, 2019-03-26:
ok so revisiting again and wanting to help here are some objectives

1) moving to gitlab so we can use the CI stuff and mainly doxygen to generate docs from source...
- we can move main "origin"  there and "push updates" to gitthub

2) I'll finish up the new www server (last step is the javascript. sigh!)

2++) remove sqlite, and move to yaml, toml, csv whatever is easier....
- we are assuming this is in a lab, so maybe login can be easier or not at all (eg localhost)

3) Generate doxygen from source files 

4) Make openplcproject.com in markdown as documentation and publish on openplcproject.gitlab.com = same

pedromorgan, 2019-03-26:
Step #5..
Setup a testing framework, so as to use upstream libs + versions..eg modbus and get problem matrix

thiagoralves, 2019-03-26:
I don’t have anything in mind, you can use csv, json, whatever you feel like is better.

pedromorgan, 2019-03-26:
what u mentioned before is "most use csv" is my curious...
ie 
- a heading and col defs
- and rows of data

pedromorgan, 2019-03-26:
I kinda want to split out stuff.. into seperate repos..

eg a "platforms" repos
- and its assosiated platform as directories and binding..

This will make CI testing much easier

pedromorgan, 2019-03-26:
MIDI = Music digital interface...(should be easy to attach sound and alike to events ? kick snare etc)
there is a ridim in a machine
So we should be able to "synthesize" a machine and sound.  eg...
(trust me, am 53+ years and hot engineer with a s.600 sapler, and sbx80. timecode machine)

garretfick, 2019-03-29:
Looks like the www server is coming back alive. I'm running into a few things with the existing www server that are making me less efficient and wanted to know if it is worthwhile addressing or just wait.

I know I'm new to this project (and so my thoughts should be appropriately weighted as near zero), but I would prioritize on splitting things up rather than moving Gitlab. You can generate docs, do tests, etc, all with Gitlab.

pedromorgan, 2019-03-29:
I get the felling I might the temp PA and coder for now..

But indeed a list o stuff to do..

Splitting up the project is what u want to do..
and the ONLY ONLY reason for gitlab.. is the pages/
and in CI can knock that stuff API  and alike as a "job" on a VM

If u are new o this realm, as i but some other experience, then cool...

>

garretfick, 2019-04-04:
I think you might have replied to the wrong person :)

pedromorgan, 2019-04-07:
@garretfick unfortunately notes to onself.. 

My PA temp is gitlab-ci and alike ..
https://gitlab.com/openplcproject
https://openplcproject.gitlab.io/openplc_v3/

If we can get cmake to gen the doxygen then cool ;-)

pedromorgan, 2019-04-07:
https://gitter.im/openPLCProjectx/community

pedromorgan, 2019-04-26:
see #59 for server in golang instead ;-)

pedromorgan, 2019-07-25:
aww #69

closed 2019-07-25 by pedromorgan