thiagoralves/OpenPLC_v3 #38: IEC std lib definition of tm does not match the standard C++ definitino

garretfick, 2019-04-01:
I'm trying to fix some issues with DNP3 and in the process was trying to get things to compile locally on my machine with Visual Studio (yes, I know, I'm doing it the hard way).

In any case, I noticed that the tm struct (iec_std_lib.h, line 275) does not match the standard definition from C90 (see http://www.cplusplus.com/reference/ctime/tm/).

Is there a particular reason for this difference? It is part of the IEC standard?

The reason I'm asking is I'm running in definition errors and trying to figure out what is the best way to resolve them.

thiagoralves, 2019-04-08:
This file is part of MatIEC, which is the compiler that generates C code from ST code. I don't know why they defined tm like that, I haven't gone much deep on MatIEC. Anyway, trying to compile this thing on Visual Studio will be a pain. If you want to compile natively on Windows try at least using MinGW

garretfick, 2019-04-08:
No problem. I was able to get things to compile with some tricks up my sleeve using the VC compilers. As before, I'm hoping to share this by the end of the week.