thiagoralves/OpenPLC_v3 #39: Unit testing of OpenPLC C++ code

garretfick, 2019-04-01:
Hi,

I'm working on making some changes to the C++ code for DNP3. I've noticed that there are no unit tests for any of the C++ code (or at least as far as I can tell).

I'd like to propose the following:

catch2 as a test framework
CMake as a build system for tests

I haven't used catch2, but it seems to be a more popular than google test (the last C++ testing framework I used).

I have some mixed feelings on this proposal - CMake is a heavy dependency, however, that dependency is already there from OpenDNP3. It is also very possible that this would only be necessary for tests.

Any thoughts on these approaches?

pedromorgan, 2019-04-07:
@garretfick canu help

I code in c/c++ (almost rare nowadays)..

so can u help with cmaking olde libs in this repos ?

certainly interested in the cmake exe gen and .deb .msi etc

garretfick, 2019-04-08:
Great - I've got a good deal of experience in C/C++, so there are no language challenges here for me.
I've made some progress on this and I'm working to make it shareable. I have a suite of tests for the DNP3 code that pass.

My plan for this is to address some issues with DNP3 and creating this as a way to ensure backward compatibility. Expect to see something this week.

@pedromorgan What did you mean by "so can u help with cmaking olde libs in this repos ?"

thiagoralves, 2019-04-08:
Awesome Garret! It will be nice to have a more robust DNP3 implementation

garretfick, 2019-04-19:
Since I created this, I'm going to close this. The recent work for robust DNP3 includes unit tests.

closed 2019-04-19 by garretfick