thiagoralves/OpenPLC_v3 #41: st_optimizer.cpp

pedromorgan, 2019-04-07:
So am new to .st

But all this seems to do is ??
https://github.com/thiagoralves/OpenPLC_v3/blob/master/utils/st_optimizer_src/st_optimizer.cpp

Loking at it, this would be simple few lines function in python..
def st_optimize(inp_file)

am so happy I dont code c++ anymore ;-)))))

pedromorgan, 2019-04-07:
same for glue_generator.cpp

Thinkinkg . its a template.

thiagoralves, 2019-04-08:
This is just a hack I did some time ago to fix a bug on PLCOpen Editor code generation. When your program has multiple coils in parallel, it was generating wrong ST if statements. The "st_optimizer" fixes that by putting each coil on a separate if statement. The right way to do it was to fix PLCOpen Editor code directly, but at that time I wasn't very familiar with PLCOpen Editor code and didn't want to mess up things. I believe this bug is fixed now, I'll just check to make sure. If it is, we can get rid of st_optimizer entirely.

About glue_generator, this is required as it is, because it creates the pointers to the data structures on the user program and links them with OpenPLC buffers. It must happen during compilation time.

pedromorgan, 2019-04-08:
Questions answered, thanks

closed 2019-04-08 by pedromorgan