thiagoralves/OpenPLC_v3 #43: which dnp3.cfg ?

pedromorgan, 2019-04-11:
There seems to be two files
- https://github.com/thiagoralves/OpenPLC_v3/blob/master/webserver/core/dnp3.cfg
- https://github.com/thiagoralves/OpenPLC_v3/blob/master/webserver/dnp3.cfg

Both files are different ?
so which is the correct one, or is is "generated" somewhere ??

pedromorgan, 2019-04-12:
Also on windows build,
it seems dnp3 is disabled ??
```
    echo ""
    echo "[OPEN DNP3]"
    cd webserver/core
    mv dnp3.cpp dnp3.disabled
    if [ $? -ne 0 ]; then
        echo "Error disabling OpenDNP3"
        echo "$ERROR_MESS"
        exit 1
    fi
    mv dnp3_dummy.disabled dnp3_dummy.cpp
    if [ $? -ne 0 ]; then
        echo "Error disabling OpenDNP3"
        echo "$ERROR_MESS"
        exit 1
    fi
    cd ../..
```

pedromorgan, 2019-04-12:
@thiagoralves so dnp3.cfg sorted, can u comment on the "windows" issue above ?

am confused...

thiagoralves, 2019-04-12:
DNP3 is disabled on windows because opendnp3 fails to compile under Cygwin. I tried many times with different configurations and gave up. So there is no DNP3 for windows builds.

thiagoralves, 2019-04-12:
> The snag with that is that the compile fails, as its looking for dnp3 functions .. or am i missing something

The build script should replace dnp3.cpp with a dummy file that only has the function calls but nothing implemented.

closed 2019-04-12 by pedromorgan