thiagoralves/OpenPLC_v3 #48: Workspace Discussion

pedromorgan, 2019-04-21:
One of the snags a the moment is that data is in sqlite.db,
and this makes it difficult to change etc and maintaing, update

So sql.db is what I would have gone for also..
but the only snag from experience is where one want to update shema etc and olde db.. arghh!!!

So can we introduce "workspace" or "work" directory..

This is required when u start a server...
and the idea is all your stuff in one external directory (can be gitable, dropable, zipable)

and instead of having "tables" for data,
we have files and directories instead.

```
openplc -w ~/my_plc_stuff```
```
# Suggested Layout

```/config.yaml``` - main config

pedromorgan, 2019-04-21:
Here's skel am working with and problem straightaway.. 

https://gitlab.com/openplcproject/openplc_go/blob/master/static/etc/config_skel.yaml

pedromorgan, 2019-04-21:
`/slaves` - where all the slaves data are in yaml,json, even csv

files in here would be reflects in the dir
eg `/slaves/rpi-new.yaml`

so to get list of slaves or api, means scanning all the files instead..
and this can be handy of someone is editing manually ;-))
ie we need a rereah button or a file watch..!

garretfick, 2019-04-23:
Did any work happen on this recently? It hit me today where a corrupt DB meant I couldn't start OpenPLC. I think some work here might be good.

pedromorgan, 2019-04-24:
am working on a go_version (for fun). all files stored in a workspace/*.yaml files. It also imports db.
https://gitlab.com/openplcproject/openplc_go/tree/pedro

pedromorgan, 2019-04-25:
Clising and assing to #56

closed 2019-07-25 by pedromorgan