thiagoralves/OpenPLC_v3 #53: Separate user supplied and system provided files

garretfick, 2019-04-23:
Currently, files that form the OpenPLC platform and user provided files are written to the same directory. A result of that is `git status` shows a lot of changes, and thus creating a complex history. It can also cause merge issues since they may write to the same files.

Thus, I would like to separate user supplied files from those that are part of the platform. There is already some work that moves the `core` files into `etc`. This change takes that a little further to move those files into `etc` and have user files in `usr`. In the `usr` folder, propose to have the following:

`usr\st_files` <= uploaded ST files
`usr\core` <= generated by iec2c

pedromorgan, 2019-04-24:
The scripts here generate into ./build/ and ./build/src_gen direcory, but can be easilly changes
https://github.com/thiagoralves/OpenPLC_v3/pull/46
https://github.com/thiagoralves/OpenPLC_v3/pull/46/files#diff-0414f4f9d80541a6cb52c13b4006d5fa

garretfick, 2019-04-24:
I saw that before but it looks like you started but it hasn't been completed. I'm concerned that there are lots of things happening (including my recent contributions), but not being merged and this is going to quickly create a nightmare.

closed 2019-04-26 by garretfick