thiagoralves/OpenPLC_v3 #56: Suggested Structure

pedromorgan, 2019-04-25:
So this is the way I got in on https://github.com/thiagoralves/OpenPLC_v3/pull/46

# OpenPLC_v3/*

- `install.sh`
- `start_openplc.py`
- `/etc/`
  - `core/` 
    - `*.cpp/h` - main.cpp, app and header files (dnp, modbus server etc)
    - `/hardware_layers/*.cpp` - platform files
    - `/lib/*.h` - iec_* files etc
  - `docs/*` - doxygen config+files  (also add api stuff here)
  - `lib/*.txt` -  pile of .txt files
  - `st_files/` 
    - `blank_program.st`
  - `openplc_default.db` - default db (soon to be abandoned)
- `/scripts/`
    - `common.sh` - common vars to all scripts
    - `change_hardware_layer.sh` 
    - `compile_program.sh` 
- `/utils/` - all the sources and some third party libs
 - `/webserver/` - all python and html templates only
-  `/build/` - 
    -  `bin/*` - where executables are build to
    -  `src_gen/*` - where src code is copied to and compiled

pedromorgan, 2019-04-25:
`workspace/` is a path outside openplc where data is stores as text/yaml files and is working on the goversion, inc importing olde db - [openplc_go](https://gitlab.com/openplcproject/openplc_go/tree/pedro)

```
openplc_server -o ./OpenPLC_v3 -w ./my_workspace -i ./openplc_legacy_to_import.db
```

- `workspace/`
  - `config.yaml` - main config file (eg port, address, debug)
  - `settings.yaml` 
  - `/users/*.yaml` 
  - `/programs/*.yaml` 
  - `/slaves/*.yaml` 
  - `/uploads/` - uploaded files

garretfick, 2019-04-26:
I would suggest to put the source files *.cpp/*.h in something other than /etc. etc is normally for configuration files, and these don't feel like configuration files.

Maybe the below?

* `install.sh`
* `start_openplc.py`
* `etc/`
  * `oenplc_default.db` - default db (soon to be abandoned)
  * `*.cfg` - configuration files
* `runtime/`
  * `core/`
    * *.cpp/h - main.cpp, app and header files (dnp, modbus server etc)
  * `docs/*` - doxygen config+files (also add api stuff here)
  * `lib/`
  * `hardware_layers/*.cpp` - platform files
* `scripts/`
  * `common.sh` - common vars to all scripts
  * `change_hardware_layer.sh`
  * `compile_program.sh`
* `utils/` - all the sources and some third party libs
* `webserver/` - all python and html templates only
* `build/` -
  * `bin/*` - where executables are build to
  * `src_gen/*` - where src code is copied to and compiled

pedromorgan, 2019-04-26:
Yep I'd go with that.....

What about vendor/ stuff ??

garretfick, 2019-04-26:
Through the magic of editing - modified above

garretfick, 2019-04-26:
To achieve this, I think the first would be to move scripts, and then make sure it all works. Then move other things

pedromorgan, 2019-04-26:
Yea its only a few teaks on my build_refactor path.. will integrate...

thiagoralves, 2019-04-26:
I like that idea too

pedromorgan, 2019-04-26:
@thiagoralves u ok with the struct from @garretfick at https://github.com/thiagoralves/OpenPLC_v3/issues/56#issuecomment-486905701

pedromorgan, 2019-04-26:
Here's what it looks like
https://github.com/pedromorgan/OpenPLC_v3/tree/develop_refactor

pedromorgan, 2019-07-25:
this is moot

closed 2019-07-25 by pedromorgan