thiagoralves/OpenPLC_v3 #58: cmake builds

pedromorgan, 2019-04-26:
Just notice that there's a cmake of gluegenerator.. 
so being a muppet to cmake kinda

```
mkdir build/glue_evo
cd build/glue_evo
cmake ../../utils/glue_generator_src/
``` 
and it compiled ;-))
but ended up up with 2 files
```
utils/glue_generator_src/bin/glue_generator
utils/glue_generator_src/bin/glue_generator_test
```

So it works..
Main question is whats strategic plan ??

pedromorgan, 2019-04-26:
I then run 
```
mkdir new_build_space
cd new_build_space
cmake ../OpenPLC_v3
make
```
and crashed with (umm gonna push build to gitlab)

pedromorgan, 2019-04-26:
https://gitlab.com/openplcproject/openplc_v3/pipelines

Yipee.. got loads of emails..

garretfick, 2019-07-24:
This has now been committed and is in development.

pedromorgan, 2019-07-25:
yipee :-) and thanks.. supervisor.coder.it.and.all from bottom

closed 2019-07-25 by pedromorgan