thiagoralves/OpenPLC_v3 #61: install and meet some error msg

HowardWhile, 2019-04-29:
Hi 
When I type cmd with `./install.sh linux`. on my ubuntu 16.04 pc.
 I have encountered some error messages during installation. Like the following.
``` sh
glue_generator.cpp:269:48: error: no matching function for call to ‘std::basic_ofstream<char>::basic_ofstream(std::__cxx11::string&, const openmode&)’
  ofstream glueVars(output_file_name, ios::trunc);

```
I think it might be related to the compiled parameters. 
So I replaced `g++` in install.sh file with `g++ -std=c++11`.
And it working normally.

thiagoralves, 2019-05-09:
I just fixed it on the latest commit

closed 2019-05-09 by thiagoralves

LitchiCheng, 2023-02-26:
Hi，
Now i meet the output below，please gave me a hand

[MATIEC COMPILER]
' is already registered with AC_CONFIG_FILES.
../../lib/autoconf/status.m4:288: AC_CONFIG_FILES is expanded from...
configure.ac:69: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
./background_installer.sh: line 70: ./configure: No such file or directory
make: *** No targets specified and no makefile found.  Stop.
cp: cannot stat './iec2c': No such file or directory
Error compiling MatIEC
OpenPLC was NOT installed!