thiagoralves/OpenPLC_v3 #66: Standardize message payload between web server and the runtime

garretfick, 2019-07-24:
## Problem

Currently, we pass messages between the runtime and the webserver for control operations using carefully written strings send over sockets. This has caused problems during development because either only one side of the message is updated or I would like to safely change the message. Furthermore, it is easy to "break the build" because of an incompatibility. This also makes it harder to develop a new web server since it must conform to an undocumented API.

## Solution

We will define the allowed messages between the web server and runtime using a language independent format (IDL). The candidates here are protobuf and flatbuffers. This allow you to define the messages in a special syntax and then you use a special compiler to convert that into platform specific code.

While protobuf are definitely more popular, flatbuffers may be more appropriate in this case because they a nicer to the heap which is important for embedded systems.

## Approach

The plan for this is to define the message that we want to allow. There are some improvements that would be very nice (for example querying a subset of the logs) that would also improve overall performance.

garretfick, 2019-07-25:
## Approach (Continued)

To fill in more details - I would suggest that we define the messages (the IDL part) in a separate repo. This would enable one of my other goals which is to write a new web server. That web server could depend on the OpenPLC runtime but be implemented in a separate Git repo and project until it reaches parity.

pedromorgan, 2019-07-25:
Think seperating the webserver out is a good idea, although @thiagoralves mentioned he want to "keep it all in one simple repos"..

I also started on a webserver written in golang and compiles to a single executable.. which needs testing (been busy) https://gitlab.com/openplcproject/openplc_go

garretfick, 2019-07-25:
I don't want to speak for @thiagoralves , but I think his concern has been ensuring that things don't break and user's have a straightforward experience. I definitely agree with that objective, but I think we can achieve that in multiple ways.

We have been putting in some time to improve testing (and more is coming). I'm also looking at making things deployable via multiple docker containers to help with some security goals I've got.

pedromorgan, 2019-07-25:
So issue is here.. as on chat with @garretfick 
```
It's complex - protobuf sounds nice, but it doesn't say "this string can only have 10 characters"
The web server needs to tread all incoming data as being hostile and validate the input. With jsonschema, you can define that in the schema, but with protobuf, you have to do it manually
```

pedromorgan, 2019-07-25:
Main issue here is that
1) the openplc run time is running..
2) it can send messages to orphaned client ?

garretfick, 2019-07-25:
My concern is more: can a badly behaving web client disrupt the runtime so that it cannot achieve expected realtime deadlines. I see the web server as being the piece that sits in-between the real time runtime and a malicious user and ensures that the runtime is safe.

pedromorgan, 2019-07-25:
My concert more is more.. how did I as a script kiddie, manage to bombard server with all ptortocols and bring it down...

Yep were onto that one, read @thiagoralves thesis which explains that problem...

But again using RrealWorld..
-  a local factory here with 100 staff, 
- HALT.. cos some one stuck a new cascade router in from yonks ago eg < 2000
- new plan works with structured wiring and only 20k spend.. (and intial install has paid for itself)..

pedromorgan, 2019-07-25:
any of the protocols can be "dossed".. in fact we want to do that.. like a lost client and roll/back... and disregard..

pedromorgan, 2019-07-25:
However, isnst it all a LAN at the end of the day ?? eg a windfarm ?

garretfick, 2019-07-25:
These are small devices that cannot withstand a DOS - easy to take down. But there are other scenarios.

I once had an intern who accidentally did a DOS against ourselves.

How we might address this would be when the clients asks for logs, the web server caches and only retrieves a diff from the runtime. Or it limits the rate at which modbus can start and stop, again protecting from that perspective. (I'm sure there are more as I've only put a basic amount of thought into this)

pedromorgan, 2019-07-25:
so being a muppet.. new person..
Things would fails cos I address 127.0.0.1
however what I need is a subnet address to fix
eg 192.168.1.53

So the person allows 0.0.0.0 vos it works..

So @garretfick u got a good argument there...

Maybe we need 2 security levels
- hacker
- production
- minitoring

garretfick, 2019-08-06:
I've done a bit of research here - I think flat buffers might be the best approach. Apparently it doesn't pack as nicely as Protobuf, but it is known to work on some systems that don't have an OS (which is something I'd like to see possible in the future).

garretfick, 2019-12-19:
I've done some more research on flat buffers. Unfortunately, the implementation makes using Python difficult.  At this point, if this is important, I need to find another solution.

pedromorgan, 2019-12-19:
What about protobuf? or is that out as well ?

garretfick, 2019-12-19:
It isn't out and I've also used it before. I'm not planning on doing anything on this in the immediate future. There are a few and part of the question is best fit for purpose.