thiagoralves/OpenPLC_v3 #70: Apply a code-style check as a part of the automated build

garretfick, 2019-07-25:
## Problem

There are several developers who are contributing code to this project. We each have our own preferences for how things are structured, but that's bad. We should have a common style to make the code easier to read, to eliminate potential bugs, and improve overall quality.

## Solution

Improve the CI build scripts to include automated code quality checks using domain-specific style checkers. For the C++ code, this would be cppcheck as a start.

pedromorgan, 2019-07-25:
This will work also as were using doxygen to create api docs so they should marry also

garretfick, 2019-12-12:
https://sonarcloud.io/ might be a good choice. I've experimented with a codacy, but I get web service problems regularly.