thiagoralves/OpenPLC_v3 #73: workspace/ dir and problems

pedromorgan, 2019-07-27:
so looking at recent `development branch` there are a couple of issues I wish to raise...

The root of the problem/snag is that the build process rewrites files within the repos itself, eg the openplc.db file. I only truly realised this problem when i started refactoring the www server.

The recent restructure with the 'runtime/' , `scripts` and cmake etc is cool, but doesnt solve the whole problem. Its also confusing for a new dev like myself.

To really solve it, I think there needs to be a "openplc_workspace/", play/, my_stuff/ or whatever its called outside of openplc file structure completely.. In fact the idea imho is that its a git.repo  itself for versioning as project.

so the structure with the "workspace" dir could have

/bin/ = all binary exe's in here
/build = cmake compile and stuff
/config/ = opelplc_platform, hardware, curr_program // etc
/db/* = all yaml files replacing the db
/etc/* = other stuff
/vendor/* = thirdparty stuff
/secrets/* = ssh keys etc

This means all the "scripts" + www etc need to have a "workspace" config somehow somewhere...

garretfick, 2019-07-31:
I've spent a few more days working on this that might address your issues. I think the objective here should be that all user items go into /etc. I usually try to do smaller incremental changes, so not trying to address everything in one go.

Once of the big things I've done now is that nearly all the user files (except the DB and hardware_layer) are now in /etc. That includes the generated C/CPP files, ST files, etc.

The things that remain to move are the user's selected hardware layer and as you mentioned the DB.  Before we do that, I think we should brainstorm a bit on that do that changing the HW layer doesn't just overwrite the user's prior data.

I think we should treat these as two separate issues, particularly as I have a number of thoughts on DB replacement with YAML (which I'm now on board with).

garretfick, 2019-11-09:
HI Pedro - wanted to check in with you if this is still an issue.

pedromorgan, 2019-11-09:
no.. seems u fixed issues... will close

closed 2019-11-09 by pedromorgan