thiagoralves/OpenPLC_v3 #74: Build Error on RaspberryPi

mtbadakhshan, 2019-07-30:
Hello, 
I got the following error, when trying to install OpenPLC on raspberry pi.
``
g++ -g -Wall -Wpointer-arith -Wwrite-strings -Wno-unused -DHGVERSION="\"\""    -o iec2c main.o stage1_2/libstage1_2.a stage3/libstage3.a stage4/generate_c/libstage4_c.a absyntax/libabsyntax.a absyntax_utils/libabsyntax_utils.a  
/usr/bin/ld: stage3/libstage3.a(stage3.o): in function `case_elements_check(symbol_c*)':
/home/pi/OpenPLC_v3/utils/matiec_src/stage3/stage3.cc:134: undefined reference to `case_elements_check_c::case_elements_check_c(symbol_c*)'
/usr/bin/ld: /home/pi/OpenPLC_v3/utils/matiec_src/stage3/stage3.cc:136: undefined reference to `case_elements_check_c::get_error_count()'
/usr/bin/ld: /home/pi/OpenPLC_v3/utils/matiec_src/stage3/stage3.cc:134: undefined reference to `case_elements_check_c::~case_elements_check_c()'
/usr/bin/ld: /home/pi/OpenPLC_v3/utils/matiec_src/stage3/stage3.cc:134: undefined reference to `case_elements_check_c::~case_elements_check_c()'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:453: iec2c] Error 1
make[1]: Leaving directory '/home/pi/OpenPLC_v3/utils/matiec_src'
make: *** [Makefile:494: all-recursive] Error 1
``

garretfick, 2019-07-31:
Can you give some details on things you did before you got that error?

thiagoralves, 2019-07-31:
Try installing again on a fresh Raspbian image. I believe other libraries or packages might be affecting the build process

closed 2019-12-02 by thiagoralves