thiagoralves/OpenPLC_v3 #80: [development] Default settings

pedromorgan, 2019-08-10:
In refactoring, bumped into issue where "my settings" in DB here didn't have `enip` etc..

so refactoring with `defaultSettings` in py to keep running..

# Main Question ?
is there somewhere we can save the default settings, and machine readable for c/c++ etc and py et all

- Snag is that with the db, its in the db and query crashes,,,
- prefer a `runtime/etc/settings.ini` or whatever format work on most plats instead of having this problem
- eg a txt file with key=val

```py
defaultSettings = dict(
    modbus_enabled="false",
    modbus_port=502,

    dnp3_enabled="false",
    dnp3_port=20000,

    enip_enabled="false",
    enip_port=20000,

    Pstorage_enabled="false",
    Pstorage_polling=10,

    Start_run_mode="false",
    Slave_polling=100,
    Slave_timeout=1000
)
```

pedromorgan, 2019-08-10:
Also needs to have the "sensible" defaults, and correct ones, for resetting..
eg 
modbus_port=123
dnp3_port = 500

pedromorgan, 2019-08-10:
Note the default button on `enip` which is the possible

[image]

pedromorgan, 2019-08-10:
key=val pair problem

garretfick, 2019-11-09:
Hi Pedro,

I've been working on a change so that it is possible to run and configure OpenPLC without the web front end. I have it working now for DNP3 and have a bit more testing to do for persistent storage. I plan to look at the Modbus master afterward

It might be good to sync-up on this issue as I think my changes will affect how you might implement this.

innir, 2019-12-02:
Isn't that what @garretfick implemented in https://github.com/thiagoralves/OpenPLC_v3/blob/development/config.example.ini ?

garretfick, 2019-12-03:
It sure is :)