thiagoralves/OpenPLC_v3 #99: Port to python3

innir, 2019-11-20:
Hi,

it seems to be quite trivial to port OpenPLC to Python3. I ran `2to3` and there where only some brackets added to `print` statements and the like. The only real changes to be made where removing lines
https://github.com/thiagoralves/OpenPLC_v3/blob/master/webserver/webserver.py#L2183-L2184
as Python3 is UTF-8 anyway.
Should I create a PR?

thiagoralves, 2019-11-20:
The webserver on the runtime is simple to port, as I wrote it thinking
about Py3. The problem is with the Editor. Its codebase is quite old and
rely on old libraries. If I port the runtime to Py3, it might generate
incompatibilities on systems that need to run the Editor in Py2 at the same
machine. That being said, feel free to create the PR. Just make sure that
all the dependencies on the installer (background_installer.sh) are changed
appropriately. Thanks!

innir, 2019-11-20:
@thiagoralves I created a PR against the development branch now: https://github.com/thiagoralves/OpenPLC_v3/pull/100.

I don't think using Python 3 for the runtime does any harm for the editor, even if run on the same machine as the don't directly exchange Python data structures.

innir, 2019-11-21:
@thiagoralves Sorry, the PR was not complete :-(

In Python 3 `socket.send` accepts only bytes and `subprocess.Popen` seems to behave differently as well.

I'll follow up with a new PR

innir, 2019-11-21:
This should do the trick: #102

garretfick, 2019-11-29:
@innir I think this is now merged. Can you confirm so that this can be closed?

closed 2019-11-29 by innir