thiagoralves/OpenPLC_v3 #116: Allow overriding the location of the config file path

garretfick, 2019-12-03:
I am currently working on developing integration tests for the OpenPLC runtime. My approach is to define a number of different ST files to use, then use cmake to build the project and finally run the runtime with some related configuration and a test written in Python.

In order to achieve this, I need the ability to specify, at the command line, an alternative configuration file so that I don't need to copy files around. This would also increase the flexibility for the runtime when testing.

Currently, we don't allow any command line arguments. I would like to see a general, cross platform (i.e works on Linux and Windows without mingw/cygwin) library for handing command line arguments. (This means, I exclude POSIX getopt as not cross platform.)

garretfick, 2019-12-04:
I've looked (but not tried) a few options, primarily based on https://attractivechaos.wordpress.com/2018/08/31/a-survey-of-argument-parsing-libraries-in-c-c/
and https://github.com/fffaraz/awesome-cpp/blob/master/README.md#cli
Things I was generally looking for were available as a single header file.

https://github.com/CLIUtils/CLI11 <-- maybe thee best choice
https://github.com/jarro2783/cxxopts <- would like to avoid due to regular expressions
http://tclap.sourceforge.net/ <-- still on sourceforge makes me suspicious
https://github.com/Taywee/args <-- in maintenance

CLI11 seems to be well maintained and I'm inclined to give it a try.

Any preferences before I go down that path?

innir, 2019-12-04:
I'd throw in https://github.com/gflags/gflags (never used it myself) and wouldn't rule out `tclap`, it seems to be well established.
I'd love to see a library used that's already packaged for Debian as that would it easier to use for me.

OTOH: Isn't parsing raw `argc` and `argv` just enough for now as it's only about the config file? One could have a look at a proper library if the flexibility is really needed.

garretfick, 2019-12-04:
I've looked at gflags myself but it seems more complex to integrate (a lot of projects are header only). For now, our way of integration has been directly copying into this project, so packing for Debian shouldn't be an issue. Might be nice in the future to either use submodules or conan, but that's a bit further off in the future.

Yes, just parsing argc and argv is definitely sufficient, and there's a good chance that this will be my first pass.

thiagoralves, 2019-12-04:
I vote for just simple argc argv parsing, unless we go crazy on command line arguments. I want to avoid adding a lot of external libraries since it makes it more complicated to maintain the project.

garretfick, 2019-12-04:
Thanks for the feedback. I'll go the simple route.

garretfick, 2019-12-09:
This was implemented in https://github.com/thiagoralves/OpenPLC_v3/pull/118

closed 2019-12-09 by garretfick