thiagoralves/OpenPLC_v3 #119: Ensure that licensing is "clearlydefined"

garretfick, 2019-12-09:
As an enterprise user of OpenPLC, I want to ensure that the licensing is clearly defined. There is a new initiative regarding licensing best practices.

I would like to ensure that we meet the recommendations at https://clearlydefined.io/

LeSpocky, 2021-02-16:
Maybe the hints from the [REUSE](https://reuse.software/) project by the FSFE could help, too?

Seriously: what corporate users look for first, is how the thing is licensed and if that allows usage. Missing license information prevents from usage at all.

In this case with several external subprojects (in folder `utils`) those have to be handled as well.

akohlsmith, 2025-05-05:
Any update to this ticket? It's kind of important for any potential use outside of hobbyiests.

thiagoralves, 2025-05-05:
License is GPLv3 as stated in the LICENSE file at the root of the project (master branch). I don't know if there is any more clarifying to this. If there is, please let me know.

akohlsmith, 2025-05-05:
I do understand GPLv3 and there were some other comments elsewhere (can't find them now, I'm guessing in the forum) describing how the "artifact" of the user's compiled ladder logic did *not* fall under GPLv3 even though it was created with the GPLv3'd OpenPLC Editor (or matiec, which is itself GPLv3).

And since OpenPLC is GPLv3, if I were to write a hardware layer for a custom board I designed, then that hardware layer code would be required to itself be GPLv3 since it's directly built into the runtime. I know this because I'm reasonably familiar with open source licenses, but a 700 line text document written in legalese doesn't help make things clear for anyone but the lawyers.

Where things get a little confusing is in the `webserver/lib` and `webserver/core/psm` subdirectories. All of the text files under the former explicitly state that they're LGPL licensed, but there is no license information in the files under the latter.

It is therefore reasonable to assume that if you are adding or changing code to the runtime itself, the editor or the python subsystem with the intent to distribute the changes then those changes fall under GPLv3 and must have their source available, but the ladder logic/etc. you write which simply uses the function blocks in their unaltered form then the code using those does not fall under GPLv3 (as the blocks are LGPL licensed).

However for custom python code/drivers/etc. using psm, things feel ambiguous. I would assume that custom python code being used through psm does not fall under GPLv3, but that's kind of the point of this issue. :-)

A brief blurb toward the end of the README.md which states something like this in plain english, and requests that people reach out if there are any further questions regarding the licensing would help considerably, while leaving the LICENSE file (and perhaps renaming `COPYING.LESSER` to `LICENSE` in `webserver/lib`) would keep the corporate legal types happy. This is my attempt at an example of what would make things much clearer, while leaving the detail work in the LICENSE files for the lawyers to argue over:

## Licensing TL;DR
The OpenPLC core, runtime, psm and editor are GPLv3 licensed. This means that you are free to use and distribute binary code utilizing the unaltered source files as long as your attribute the use of OpenPLC and make the OpenPLC source code available. If you make changes to the code in these components and distribute binaries based on these changes then you must also make your changes to the source code available.

The OpenPLC function blocks located `webserver/lib` are LGPL licensed; this means that you are free to use these in your own code without changes, and you may distribute any code which uses these function blocks freely and without obligation to make the source code to your own programs using these function blocks available. If, however, you make changes to these function blocks then you must make the source code to the function blocks (including any changes) available.

The source code to any Python scripts which use the OpenPLC psm layer does not have to be disclosed, but if you make changes to the psm layer itself, then those changes must be published.

This arrangement allows you to use OpenPLC freely and without requirement to publish the source code to your own PLC programs and functions (including custom Python blocks), while ensuring that any changes you make to the OpenPLC project itself remain open and available to anyone.