thiagoralves/OpenPLC_v3 #128: Installing on Raspberry Pi  Kali

ansco11, 2020-02-17:
I cloned commit 2d535b8dca39a168283b1f3e83131763a3a73b26 (Jan 5th) onto a Raspberry Pi 3B+ running Kali. Kali does not have the wiringpi package, So I cloned https://github.com/WiringPi/WiringPi.git and installed, but:

/usr/bin/ld: /usr/local/lib/libwiringPi.so: undefined reference to `crypt'

So I fixed with:

diff --git a/webserver/scripts/compile_program.sh b/webserver/scripts/compile_program.sh
index a19a0bc..3185e64 100755
--- a/webserver/scripts/compile_program.sh
+++ b/webserver/scripts/compile_program.sh
@@ -105,7 +105,7 @@ elif [ "$OPENPLC_PLATFORM" = "rpi" ]; then
     echo "Generating glueVars..."
     ./glue_generator
     echo "Compiling main program..."
-    g++ -std=gnu++11 *.cpp *.o -o openplc -I ./lib -lrt -lwiringPi -lpthread -fpermissive `pkg-config --cflags --libs libmodbus` -lasiodnp3 -lasiopal -lopendnp3 -lopenpal -w
+    g++ -std=gnu++11 *.cpp *.o -o openplc -I ./lib -lrt -lwiringPi -lpthread -fpermissive `pkg-config --cflags --libs libmodbus` -lasiodnp3 -lasiopal -lopendnp3 -lopenpal -lcrypt -w
     if [ $? -ne 0 ]; then
         echo "Error compiling C files"
         echo "Compilation finished with errors!"

Will the libcrypt dependency will be a problem for Raspbian?

antony-liu, 2020-02-25:
Same issue on Raspberry Pi 3B with Ubuntu 16.04.6 LTS (GNU/Linux 4.4.38-v7+ armv7l).

thiagoralves, 2020-02-25:
The officially supported OS on the Raspberry Pi is Raspbian. Your mileage may vary if you try to  install OpenPLC on other OSes. i don’t plan on supporting other OSes in the Pi beside Raspbian.

closed 2020-02-25 by thiagoralves