thiagoralves/OpenPLC_v3 #134: Can't run "start_openplc.sh" on Docker

DaiGuard, 2020-07-04:
The following error is output when starting in docker mode.

```bash
$ docker run -it --rm --privileged -p 8080:8080 openplc:v3
./start_openplc.sh: line 3: python2: command not found
```

---

## Environment

Host machine: Ubuntu 18.04.4 LTS
Docker: version 19.03.12, build 48a66213fe
Debian image: Latest

thiagoralves, 2020-10-16:
I made a few changes to the installer. Can you test it again?

liowwenan, 2024-04-15:
Hi, I have the same problem while installing OpenPLC Runtime using Cygwin. Got any solution?

thiagoralves, 2024-09-10:
start_openplc.sh is created by the installer at the end of the installation process.

jay-jjwu, 2024-09-12:
i tried to build image based on the latest dockerfile, but it seems the 'start_openplc.sh' was not created successfully.  Any idea what could be the cause?

```
FROM debian:bullseye-20240722

COPY . /workdir

WORKDIR /workdir

RUN mkdir /persistent

VOLUME /persistent

RUN ./install.sh docker \
    && touch /persistent/mbconfig.cfg \
    && touch /persistent/persistent.file \
    && mkdir /persistent/st_files \
    && cp /workdir/webserver/openplc.db /persistent/openplc.db \
    && mv /workdir/webserver/openplc.db /workdir/webserver/openplc_default.db \
    && cp /workdir/webserver/dnp3.cfg /persistent/dnp3.cfg \
    && mv /workdir/webserver/dnp3.cfg /workdir/webserver/dnp3_default.cfg \
    && cp -r /workdir/webserver/st_files/ /persistent/st_files/ \
    && mv /workdir/webserver/st_files /workdir/webserver/st_files_default \
    && cp /workdir/webserver/active_program /persistent/active_program \
    && mv /workdir/webserver/active_program /workdir/webserver/active_program_default \
    && ln -s /persistent/mbconfig.cfg /workdir/webserver/mbconfig.cfg \
    && ln -s /persistent/persistent.file /workdir/webserver/persistent.file \
    && ln -s /persistent/openplc.db /workdir/webserver/openplc.db \
    && ln -s /persistent/dnp3.cfg /workdir/webserver/dnp3.cfg \
    && ln -s /persistent/st_files /workdir/webserver/st_files \
    && ln -s /persistent/active_program /workdir/webserver/active_program

ENTRYPOINT ["./start_openplc.sh"]
```

omedirk, 2024-11-27:
it's a line ending issue, use unix endings 
clone like this:
git -c core.eol=lf -c core.autocrlf=false clone https://github.com/thiagoralves/OpenPLC_v3.git