thiagoralves/OpenPLC_v3 #139: Update python version

u2n, 2020-09-22:
Every time install routine calls python (U20.04/deb), deprecation notice prints:

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.

thiagoralves, 2021-04-16:
OpenPLC needs a big rewrite to be compatible with Python 3. Since I don't have time to do it myself, I'm closing the issue until I find someone willing to fix it.

closed 2021-04-16 by thiagoralves

KieranTHE, 2021-04-16:
> 
> 
> OpenPLC needs a big rewrite to be compatible with Python 3. Since I don't have time to do it myself, I'm closing the issue until I find someone willing to fix it.

Hadn't seen this one. How much is there to be re written?