thiagoralves/OpenPLC_v3 #147: Problem launching blank program

kiwih, 2021-03-17:
Hello, after installing OpenPLC_v3 to a raspberry pi zero, I try to start the runtime with the blank program but it appears to hang:

The log in the browser window stops here:
```
OpenPLC Runtime starting...
Skipping configuration of Slave Devices (mbconfig.cfg file not found)
Interactive Server: Listening on port 43628
Warning: Persistent Storage file not found
Issued start_modbus() command to start on port: 502
Server: Listening on port 502
Server: waiting for new client...
Issued start_dnp3() command to start on port: 20000
DNP3 ID manager: Starting thread (0)
DNP3 ID DNP3_Server: Listening on: 0.0.0.0:20000
Issued start_enip() command to start on port: 44818
Server: Listening on port 44818
Server: waiting for new client...
```

kiwih, 2021-03-17:
As it turns out it doesn't matter? I made a program that turns I/O on and off and so on and it all works fine. I think this is a UI issue - the fact that the page never finishes loading /never refreshes to say "running" made me think that it wasn't finished when it actually was.

kiwih, 2021-03-17:
Note: sometimes it _does_ finish loading, and refreshes the page, which makes the whole thing even more confusing for the novice user.

thiagoralves, 2021-04-16:
There is a minor bug on the UI that makes it unresponsive sometimes. Clicking on any other link (like "Programs" for example) will force the UI to refresh. This does not affect the actual runtime that runs on the background.