thiagoralves/OpenPLC_v3 #151: Compilation error (modbus_master.cpp)

mefuller, 2021-08-19:
Attempting to compile from source with `./install.sh linux` on a fully-updated Fedora 34 system:
```
Compiling main program...
modbus_master.cpp: In function ‘void initializeMB()’:
modbus_master.cpp:619:36: error: too many arguments to function ‘void modbus_set_response_timeout(modbus_t*, const timeval*)’
  619 |         modbus_set_response_timeout(mb_devices[i].mb_ctx, to_sec, to_usec);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from modbus_master.cpp:31:
/usr/include/modbus/modbus.h:151:6: note: declared here
  151 | void modbus_set_response_timeout(modbus_t *ctx, const struct timeval *timeout);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Error compiling C files
Compilation finished with errors!
```

Without tearing into the (unfamiliar) code, I'm at a loss as to how to resolve this.

Thanks

thiagoralves, 2021-08-19:
The installer script was made for Debian systems (with apt-get). It will fail on any system that does not have apt package manager (which is your case). You’ll have to install all dependencies by hand and then use ./install custom to install OpenPLC

closed 2021-08-19 by thiagoralves

mefuller, 2021-08-19:
All dependencies are installed and `sudo ./install custom` leads to the same error, which does not appear to be at all related to an unmet dependency:

```
Compiling main program...
modbus_master.cpp: In function ‘void initializeMB()’:
modbus_master.cpp:619:36: error: too many arguments to function ‘void modbus_set_response_timeout(modbus_t*, const timeval*)’
  619 |         modbus_set_response_timeout(mb_devices[i].mb_ctx, to_sec, to_usec);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from modbus_master.cpp:31:
/usr/include/modbus/modbus.h:151:6: note: declared here
  151 | void modbus_set_response_timeout(modbus_t *ctx, const struct timeval *timeout);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Error compiling C files
Compilation finished with errors!
```

thiagoralves, 2021-08-19:
It appears to me that you’re using a different version of libmodbus on your system, and it is trying to compile against that, which has a different API. You must use the version of libmodbus that comes with OpenPLC. Building OpenPLC on custom systems is complicated as each system has its own set of libraries and configurations. Use a Debian-based VM to play with OpenPLC and you will have a much smoother experience.