thiagoralves/OpenPLC_v3 #173: How to make OpenPLC support OPC UA?

11235yuan, 2022-05-02:
Many ICS protocols already support encryption, such as OPC UA, Modnus/TCP Security, DNP3-SA, to improve Security. I would like to modify OpenPLCv3 to support protocols like OPC UA. Can you give me a hint? Tell me which files should I change? Or a development manual?

thiagoralves, 2022-05-02:
https://openplc.discussion.community/post/how-to-make-openplc-support-opc-ua-12342333?pid=1332525522

closed 2022-05-02 by thiagoralves

RajatDas2020, 2024-07-30:
Hi, I have downloaded this version, "Actually that commit is here:  https://github.com/thiagoralves/OpenPLC_v3/pull/16" . Now in the openplc runtime there is a option of opcua server. Now I have node-red opcua client, and I want to connect this client to opcua server of the openplc. How to do that ?