thiagoralves/OpenPLC_v3 #175: access path

choukri-m, 2022-08-10:
I noticed that matiec does not seem to work when trying to use VAR_ACCESS as defined in the IEC 61131-3, section 2.7.1
This makes this implementation not fully compliant to the standard
This feature will help improve the flexibility of accessing variables externally and not rely solely on the location method
I hope this is moved forward

JohnnyOpcode, 2022-08-10:
Interesting point. I would love to start creating OOP classes with matiec.

Has anyone tried using OpenPLC to do this (add methods to Function Blocks)?

garretfick, 2022-08-10:
Could you provide an example that does not work?

choukri-m, 2022-08-11:
The limitation of using location keywords is the limit on types that can be passed. i.e. single bit, byte, word, double and long. So if you want to pass for instance a real variable, you would have to convert to one of the 5 types supported which leads to losing some precision.

choukri-m, 2022-08-11:
Please see below an example where I need to access a REAL variable:
`PROGRAM Demo
  VAR_INPUT
    In1 : REAL;
  END_VAR

  VAR
    Threshold : REAL := 1234.9876;
  END_VAR

  VAR_OUTPUT
    Out : BOOL;
  END_VAR

  IF In1 > Threshold THEN
    Out := TRUE;
  ELSE
    Out := FALSE;
  END_IF;
    
END_PROGRAM

CONFIGURATION Config0
  RESOURCE Res0 ON PLC
    TASK task0(INTERVAL := T#500ms,PRIORITY := 0);
    PROGRAM P1 WITH task0 : Demo;
  END_RESOURCE
  VAR_ACCESS
    FLOW : Res0.P1.In1 : REAL READ_WRITE;
    ALARM : Res0.P1.Out : BOOL READ_ONLY;
  END_VAR
END_CONFIGURATION`

if you run this it gives the error: `unexpected token after resource declaration`

thiagoralves, 2022-08-11:
You don't need VAR_ACCESS to access a REAL variable. VAR_ACCESS is used to
access a variable that exists on a separate program, that's why it is
called an "access path". The way you used VAR_ACCESS in this example is
incorrect. You can only declare it inside a program (which is not the case
in the example). If you need to use a REAL variable, just declare it in
your program as you did. OpenPLC (and MatIEC) supports several different
datatypes, including REAL. If you have multiple programs and you need to
share variables among them, just use GLOBAL variables instead.

choukri-m, 2022-08-11:
I want to access this variable not on openPLC but on a separate app. So basically I would like to set up a project where the soft PLC gets an input (of the type REAL) from another software without having to re-implement the software on the PLC. hence the need for use of ACCESS_VAR to basically communicate outside of the PLC

thiagoralves, 2022-08-11:
ACCESS_VAR does not communicate externally. It can only be used to create a
communication path between different programs within the PLC framework. If
you want to get input from an external source you will have to use one of
the SCADA protocols OpenPLC supports. Modbus is the most recommended one.
The only catch is that if you want to send/receive a REAL variable you will
have to concatenate two Modbus registers as REAL variables are 32-bit and a
Modbus register is 16-bit.

closed 2022-09-16 by thiagoralves