thiagoralves/OpenPLC_v3 #177: No JSON object could be decoded

taffarel55, 2022-08-21:
Hello, I followed the [example](https://openplcproject.com/docs/3-2-creating-your-first-project-on-openplc-editor/) of the site and everything worked perfectly running locally. To upload to Arduino I had the following problem:

[image]

Below is the error log:

```
app-title:
beremiz

app-version:
2.01
Release: 2022-08-17

cwd:
/home/dbserver/Downloads/OpenPLC Editor for Linux/OpenPLC_Editor

date:
Sun Aug 21 01:08:32 2022

e-type:
<type 'exceptions.ValueError'>

e-value:
No JSON object could be decoded

locals:
    s: ''
    self: <json.decoder.JSONDecoder object at 0x7f7e94143a90>
    idx: 0

platform:
Linux-5.15.0-46-generic-x86_64-with-Ubuntu-20.04-focal

python-version:
2.7.18

self:
    parse_object: <function JSONObject at 0x7f7e6590a750>
    parse_constant: <built-in method __getitem__ of dict object at 0x7f7e6590d170>
    scan_once: <_json.Scanner object at 0x7f7e659092f0>
    parse_float: <type 'float'>
    object_hook: None
    encoding: None
    parse_int: <type 'int'>
    parse_string: <built-in function scanstring>
    strict: True
    parse_array: <function JSONArray at 0x7f7e6590a7d0>
    object_pairs_hook: None

traceback:
  File "/home/dbserver/Downloads/OpenPLC Editor for Linux/OpenPLC_Editor/editor/BeremizIDE.py", line 852, in OnMenu
    getattr(self.CTR, method)()
  File "/home/dbserver/Downloads/OpenPLC Editor for Linux/OpenPLC_Editor/editor/ProjectController.py", line 2087, in _generateArduino
    dialog = ArduinoUploadDialog.ArduinoUploadDialog(self.AppFrame, program)
  File "/home/dbserver/Downloads/OpenPLC Editor for Linux/OpenPLC_Editor/editor/dialogs/ArduinoUploadDialog.py", line 236, in __init__
    self.loadSettings()
  File "/home/dbserver/Downloads/OpenPLC Editor for Linux/OpenPLC_Editor/editor/dialogs/ArduinoUploadDialog.py", line 467, in loadSettings
    settings = json.loads(jsonStr)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
<type 'exceptions.ValueError'>: No JSON object could be decoded

wx-platform:
__WXGTK__

wx-version:
3.0.2.0
```

I searched the open issues and didn't find this error.

thiagoralves, 2022-09-16:
This is not an issue for OpenPLC Runtime, but for OpenPLC Editor. In any case, this is probably a corrupted installation, as it is missing the JSON configuration file. Just reinstall the Editor and it should work just fine

closed 2022-09-16 by thiagoralves

thiagoralves, 2022-10-11:
It looks like you have a corrupted install. Reinstall the editor and make
sure that the first time you upload to Arduino you wait until it finishes
downloading all the required files. It takes a while for it to setup the
environment. Once everything is put in place, uploads will take only a few
seconds.

taffarel55, 2022-10-11:
Ok, thanks.