thiagoralves/OpenPLC_v3 #178: some bugs

kkangd, 2022-09-16:
[K-ST_A_Formal_Executable_Semantics_of_PLC_Structur.pdf](https://github.com/thiagoralves/OpenPLC_v3/files/9580460/K-ST_A_Formal_Executable_Semantics_of_PLC_Structur.pdf)

jancumps, 2022-09-16:
Is it copyrighted material? Should it be a link to the original report vs upload to github?

thiagoralves, 2022-09-16:
@kkangd Thank you for this incredible and detailed work. I appreciate you sending this over, but the issues you presented on your paper are actually related to MatIEC. Both Beremiz and OpenPLC uses the MatIEC compiler to transpile Structured Text code into C. The differences in behavior is just because the built-in MatIEC on OpenPLC Runtime hasn't been updated in a while. If you just update MatIEC in your local copy of OpenPLC Runtime it should behave exactly the same as Beremiz or any other project that relies on MatIEC.

I'm not an active contributor to MatIEC (I think I only opened one PR on it once). So perhaps it would be more useful for you to present your findings to the MatIEC guys at https://bitbucket.org/mjsousa/matiec_git/src/master/

closed 2022-09-20 by thiagoralves