thiagoralves/OpenPLC_v3 #198: Install error in windows linux wsl2

LitchiCheng, 2023-02-26:
Hi，
Now i meet the output below，please gave me a hand

[MATIEC COMPILER]
' is already registered with AC_CONFIG_FILES.
../../lib/autoconf/status.m4:288: AC_CONFIG_FILES is expanded from...
configure.ac:69: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
./background_installer.sh: line 70: ./configure: No such file or directory
make: *** No targets specified and no makefile found. Stop.
cp: cannot stat './iec2c': No such file or directory
Error compiling MatIEC
OpenPLC was NOT installed!

Littlebai12138, 2023-06-13:
Me too. Does anyone know how to solve it?

thiagoralves, 2023-06-13:
There is a Windows installer for the runtime. Use that instead

closed 2023-06-13 by thiagoralves