thiagoralves/OpenPLC_v3 #220: Unable to Start PLC when running runtime within docker under WSL2

GT-AI-UK, 2023-11-12:
First thing's first, this project is great!

I followed the installation instructions - cloned the repo, ran `docker build -t openplc:v3 .`, then started the container with `docker run -p 502:502 -p 8080:8080 openplc:v3`.

I can browse to the webui without issue and login with the default credentials.
Hardware is set to Blank Linux, which I assume is correct.
I can upload a program.
When I try to Start the PLC, the log in the webui briefly displays it's usual start up logs, until "warning: persistent storage file not found", where the log quickly vanishes. The PLC fails to start.

Docker logs below. I'm unsure exactly what the cause of the issue is, but the bottom 2 lines are particularly concerning and I believe this is where the fault lies. Regretfully, I have no idea how to troubleshoot this. I'm happy to try more things out but would require some advise as I don't know what to do!

`
OpenPLC Runtime starting...
Interactive Server: Listening on port 43628
Interactive Server: waiting for new client...
Warning: Persistent Storage file not found
Setting main thread priority to RT
WARNING: Failed to set main thread to real-time priority
Locking main thread memory
Getting current time
Interactive Server: Client accepted! Creating thread for the new client ID: 3...
Interactive Server: waiting for new client...
Interactive Server: Thread created for client ID: 3
Issued runtime_logs() command
Interactive Server: client ID: 3 has closed the connection
Terminating interactive server connections
Interactive Server: Client accepted! Creating thread for the new client ID: 3...
Interactive Server: waiting for new client...
Interactive Server: Thread created for client ID: 3
Issued start_modbus() command to start on port: 502
Server: Listening on port 502
Server: waiting for new client...
Interactive Server: client ID: 3 has closed the connection
Terminating interactive server connections
Interactive Server: Client accepted! Creating thread for the new client ID: 3...
Interactive Server: waiting for new client...
Interactive Server: Thread created for client ID: 3
Issued start_dnp3() command to start on port: 20000
Interactive Server: client ID: 3 has closed the connection
Terminating interactive server connections
DNP3 ID manager: Starting thread (0)
channel state change: OPENING
terminate called after throwing an instance of 'St9bad_alloc'
  what():  std::bad_alloc
`

thiagoralves, 2023-11-12:
Disable DNP3 on settings and try again

GT-AI-UK, 2023-11-12:
"DNP3 ID manager: Starting thread (0)" - I should have figured this is where the problem started!
Thanks so much for such a prompt reply. I've been watching your videos today and learning loads.
Thanks for making such a great tool, and for your kind assistance. Closing <3

closed 2023-11-12 by GT-AI-UK