thiagoralves/OpenPLC_v3 #224: (Stable) releases

serpent213, 2024-01-23:
Just a question, I can't find any revisions tagged as stable, the development branch is not in use any more apparently.

Is latest master always expected to be stable? For a PLC runtime that's quite an important point, I dare say.  🙂

thiagoralves, 2024-01-23:
Yes, master is always the stable release. After the migration of the Python 3 branch to the master branch, development on this editor has slowed down as the attention is now focused on the new editor: https://github.com/Autonomy-Logic/openplc-editor

closed 2024-01-23 by thiagoralves

thiagoralves, 2024-01-23:
The new runtime is pretty much a work in progress. It is not usable at all right now, and most of the development has been on the webserver architecture.