thiagoralves/OpenPLC_v3 #245: Hardware Layer compile issues with PiXtend V2 L

David-Kopczynski, 2024-07-13:
### Issue description

When trying to select the hardware layer for the PiXtendV2, you will encounter the error "Compilation finished with errors!". This issue has been mentioned in the [community forum](https://openplc.discussion.community/post/solved-error-when-compiling-unipiv1-1-hardware-layer-13187548?highlight=wiringpi%20undefined%20reference&trail=15). Looking at this codebase, this is also likely a problem for Fischertechnik, PiXtend, (PiXtend 2l), PiXtend 2s, Raspberry Pi (old), and the UniPi.

### Steps to reproduce

Try selecting one of the mentioned hardware layers.

### Temporary fix

It is possible to replace `lpigpio` with `lwiringPi` within `./webserver/scripts/compile_program.sh`.

For anyone interested, my `Dockerfile` for the PiXtend V2 L will install and patch the compilation to get all things working:

```
FROM ubuntu:latest

ARG TZ
ENV TZ=$TZ

# Configure timezone
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

WORKDIR /app

# Install Raspberry Pi/PiXtend hardware support with dependencies
RUN apt update && \
    apt install -y git sudo make build-essential

RUN git clone https://github.com/WiringPi/WiringPi.git && \
    cd WiringPi && \
    ./build && \
    cd .. && \
    rm -rf WiringPi

# Install OpenPLC Runtime
# https://autonomylogic.com/docs/installing-openplc-runtime-on-linux-systems/
RUN apt-get update && \
    apt-get install -y unzip python3 cmake wget

RUN git clone https://github.com/thiagoralves/OpenPLC_v3.git . && \
    ./install.sh rpi

# Copy initial configuration
COPY ./pixtend /app/webserver/

# Set PiXtend V2 L as default hardware
RUN cd ./webserver/core && \
    ../scripts/change_hardware_layer.sh pixtend_2l && \
    cd ../..

# Fix build script for PiXtend
# https://openplc.discussion.community/post/solved-error-when-compiling-unipiv1-1-hardware-layer-13187548?highlight=wiringpi%20undefined%20reference&trail=15
RUN sed -i 's/lpigpio/lwiringPi/g' ./webserver/scripts/compile_program.sh

# Compile default program
RUN cd ./webserver/scripts && \
    ./compile_program.sh blank_program.st && \
    cd ../..
```

This is also my simplified docker-compose.yml that should be able to run this image:

```
services:
  pixtend:
    container_name: pixtend
    build:
      dockerfile: ./Dockerfile
      args:
        - TZ=Europe/Berlin
    command: ./start_openplc.sh
    stop_signal: SIGKILL
    ports:
      - 80:8080
    privileged: true
    restart: unless-stopped
    tty: true
```

thiagoralves, 2025-04-23:
Could you please check if this error still occurs now that we've moved back from pigpio to wiringpi?

David-Kopczynski, 2025-05-03:
Sorry for the long wait! Sadly, I do not have a PiXtend at hand and thus forwarded this to someone to test. However, the updated `Dockerfile` without the line for `RUN sed -i 's/lpigpio/lwiringPi/g' ./webserver/scripts/compile_program.sh
` seems to work just fine!

thiagoralves, 2025-05-05:
That's great. Thanks for checking!

closed 2025-05-05 by thiagoralves