thiagoralves/OpenPLC_v3 #248: Basic walkthrough of deploying and running OpenPLCv3 missing

aspanner, 2024-07-24:
Hi Thiago, I am missing a basic walkthrough of how to build/deploy/run a program/flow on the PLC.

I am happy to work with you to create the documentation but will need your guidance.

thiagoralves, 2024-07-24:
Does this work? https://autonomylogic.com/docs/3-2-creating-your-first-project-on-openplc-editor/

aspanner, 2024-07-25:
Thanks, Thiago. It's certainly a great start. However customers want to run PLCs on OpenShift, MicroShift and Podman. 
How can we test proper deployment without wiring?

thiagoralves, 2024-07-25:
OpenPLC was not designed for that.

aspanner, 2024-07-25:
Do you mean not designed for testing without wiring or not designed for running on a container platform?
Can we leverage Ethercat to test?

thiagoralves, 2024-07-25:
Not designed for containers (although there is a docker image created by the community), hence there is no documentation for that. Wiring or IO of any kind is not required, although running a PLC without IO makes no sense to me. You can use any external Modbus TCP or RTU as IO. The documentation to add external Modbus IO is here: https://autonomylogic.com/docs/2-6-slave-devices/

Also, you can still follow the walkthrough I sent you and just skip the wiring part. You’ll be able to upload the program, but won’t be able to see it doing anything.

aspanner, 2024-07-25:
Log messages as step 1 would suffice.
I agree with you, then I will need  IO.
What I don't understand though is why you say it would not work even though TCP based IO is supported?

thiagoralves, 2024-07-25:
I didn't say it won't work, just that you won't be able to see or do anything unless you attach some IO. You gotta have something, even if it's just virtual IO like Modbus slave simulators attached. The other option is to debug the running program from the editor (blue bug icon from the toolbar on the editor). On the debugger you can see the diagram running live and can force inputs/outputs to trigger events. That's also another way to "see" some action.

closed 2024-07-31 by thiagoralves