thiagoralves/OpenPLC_v3 #253: Modbus should accept multiple ADUs per TCP frame

Abestanis, 2024-08-27:
I have a Modbus client that is sending multiple commands with the click of a button to the Modbus server of OpenPLC. Sometimes, some of the last Modbus commands are not executed by OpenPLC.

### Details
According to the [Modbus TCP specification](https://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf), a Modbus message is the combination of exactly one ADU per TCP frame, which contains exactly one PDU.
<details><summary>Screenshots from the spec</summary>

[image]
[image]
</details> 

My client is a simple nodejs program that runs on a Windows machine and uses [`modbus-serial`](https://www.npmjs.com/package/modbus-serial) to communicate with the OpenPLC Modbus server. And unfortunately, my client is sometimes sending multiple APUs per TCP frame.

<details><summary>Screenshot of Wireshark capture</summary>

[image]
</details>

### Why this should be fixed in OpenPLC and not on the client
Now I know the OpenPLC is conforming to the Modbus specification and it's the client that is not conformant (there is even [an open but stagnant issue](https://github.com/yaacov/node-modbus-serial/issues/540)), but there are multiple reasons why I think this should be handled on the OpenPLC side:

1. The spec is not clear. On the one side, it specifically forbids multiple ADUs in one TCP frame:
    > A TCP frame must transport only one MODBUS ADU.
   
   But then in the next sentence it seems to allow the possibility of sending multiple "Modbus requests" in one "TCP PDU":
   > It is advised against sending multiple MODBUS requests or responses on the same TCP PDU.

   I'm pretty sure a "TCP PDU" is a TCP frame. I initially thought a "MODBUS requests" request refers to a Modbus PDU, but the possibility and framework of having multiple PDUs per ADU is mentioned nowhere else in the spec, so I believe a "MODBUS requests" should be a ADU, which seems to directly contradict the first sentence.
2. The requirement to have only one APU per TCP frame is incredible hard to implement on a system with an operating system that has a TCP stack. TCP is first and foremost a *streaming* protocol with no real notion of packet boundaries, framing is an implementation detail that is not exposed to the layer above. None of the operating systems provide enough control over the framing process to guarantee not merging frames, so the only option is to bypass the TCP stack completely and use some form of raw socket, but this means the client has to implement most parts of the TCP protocol themselves, which seems unreasonable.
3. The OpenPLC Modbus server is actually also not spec compliant because it has exactly the same issue when responding to requests.  Since the server is just using a regular socket, packets can be merged by the underlying operating system. Also, while [the `read`](https://github.com/thiagoralves/OpenPLC_v3/blob/3c08a69a6577859e14833d4b3df5974a0f356585/webserver/core/server.cpp#L174) seems to return the data one TCP frame at a time, there is actually no guarantee about that.
4. I think OpenPLC should follow the robustness principle "be conservative in what you send, be liberal in what you accept". Now we can't easily fix the sending part, but we can be more lenient in what we accept.

### Proposal
The Modbus implementation of OpenPLC should treat the incoming data as a stream and ignore TCP frame boundaries. We can use a ringbuffer to archive this. When more data becomes available, the server should try and execute multiple APUs from the buffer until there is not enough data in the buffer to parse the next Modbus APU.

I am willing to work on a PR if you agree with this path forward.

thiagoralves, 2024-08-27:
Thanks for the info @Abestanis. Feel free to open a PR to fix this behavior. As long as it keeps working with devices that follow the specification, I'm fine with it.

Abestanis, 2025-02-04:
@thiagoralves Can you please re-open this issue, since #254 has been reverted.

PR #254 (2024-12-31): Use libmodbus to decode incoming Modbus TCP messages
This allows us to better handle multiple Modbus ADUs in a single TCP frame (fixes #253).

`libmodbus` is better at parsing ADUs and we already use it when we are the Modbus client.

I tested this by sending many commands in a short amount of time and verifying via Wireshark that some of them were sent in one TCP frame. I made sure that OpenPLC was sending one response per request, even for those in one TCP packet.

In addition to the receiving problem, I also noticed that the result of `write` was ignored, which returns the number of bytes w [... cut]