thiagoralves/OpenPLC_v3 #255: win can't install

Environmental-lion, 2024-09-03:
[image]

thiagoralves, 2024-09-03:
Sorry, there seems to be a glitch on the code rendering for the website docs. The command must have two dashes for the "noconfirm" option (should be --noconfirm instead of -noconfirm). Even though the text contain two dashes, the docs page is rendering only one. I've changed the format of the display box to show the actual command that needs to be typed in. Now the instructions are correct. Please try again.

Also, did you try the automatic installer from the docs page? Didn't it work for you?

closed 2024-09-03 by thiagoralves

Environmental-lion, 2024-09-04:
The automatic installer will crash.I want to take a screenshot for you.But it crash very fast.

Environmental-lion, 2024-09-04:
[image]
[image]
It unsuccessful.

thiagoralves, 2024-09-04:
You must type one command at a time. Only the first command was processed.

Environmental-lion, 2024-09-04:
[image]
I've tried so many times.

thiagoralves, 2024-09-04:
You must be on a filtered internet which is causing the SSL certificate to fail. You have two options: 
1. move to an unfiltered internet where you can download stuff
2. try to download the OpenPLC repository straight from GitHub and then extract it inside your MSYS2 folder.