thiagoralves/OpenPLC_v3 #273: Vulnerability Found - Memory Corruption in server.cpp Due to Improper Thread Argument Handling

d00gman, 2025-04-23:
# Memory Corruption in server.cpp Due to Improper Thread Argument Handling
# Summary
There is a memory corruption and denial-of-service vulnerability in `server.cpp`, specifically in how thread arguments are passed to the `handleConnections()` function.

Currently, the arguments are passed using a stack-allocated array:

```cpp
int arguments[2];
pthread_create(&thread, NULL, handleConnections, arguments);
```

This is unsafe because the thread may access arguments after the parent stack frame is gone or overwritten, leading to invalid memory reads, corrupted logic, and crashes. This behavior has been confirmed via GDB and live testing.

# Proposed Fix
Switch to heap-allocated thread arguments using a structured type (thread_args_t) and add runtime validation:
```cpp
typedef struct {
    int magic;
    int client_fd;
    int protocol_type;
} thread_args_t;
```
Set a magic constant (e.g., `0xADE0BEDD`) for runtime validation inside `handleConnections()`.

# Proposed Patch
```diff
--- server_orig.cpp	2025-04-23 11:21:34.494949394 -0400
+++ server_fixed.cpp	2025-04-23 11:01:47.718844293 -0400
@@ -37,6 +37,13 @@
 #define MAX_MODBUS 100
 #define NET_BUFFER_SIZE 10000
 
+#define THREAD_ARGS_MAGIC 0xADE0BEDD
+
+typedef struct {
+    int magic;
+    int client_fd;
+    int protocol_type;
+} thread_args_t;;
 
 //-----------------------------------------------------------------------------
 // Verify if all errors were cleared on a socket
@@ -198,9 +205,19 @@
 void *handleConnections(void *arguments)
 {
     char log_msg[1000];
-    int *args = (int *)arguments;
-    int client_fd = args[0];
-    int protocol_type = args[1];
+    int client_fd = 0;
+    int protocol_type = 0;
+    thread_args_t *args = (thread_args_t *)calloc(1, sizeof(thread_args_t));
+            args->magic = THREAD_ARGS_MAGIC;
+            if (args->magic != THREAD_ARGS_MAGIC) {
+                sprintf(log_msg, "Server: Invalid arguments structure in handleConnections! Magic value: 0x%X\n",args->magic);
+                log(log_msg);
+            }
+            else {
+                client_fd = args->client_fd;
+                protocol_type = args->protocol_type;
+            }
+    free(args);
     unsigned char buffer[NET_BUFFER_SIZE];
     int messageSize;
     bool *run_server;
@@ -268,7 +285,7 @@
     while(*run_server)
     {
         client_fd = waitForClient(socket_fd, protocol_type); //block until a client connects
-        if (client_fd < 0)
+        if (client_fd < 0 )
         {
             sprintf(log_msg, "Server: Error accepting client!\n");
             log(log_msg);
@@ -281,12 +298,20 @@
             int ret = -1;
             sprintf(log_msg, "Server: Client accepted! Creating thread for the new client ID: %d...\n", client_fd);
             log(log_msg);
-            arguments[0] = client_fd;
-            arguments[1] = protocol_type;
-            ret = pthread_create(&thread, NULL, handleConnections, (void*)arguments);
-            if (ret==0) 
-            {
-                pthread_detach(thread);
+            thread_args_t *args = (thread_args_t *)calloc(1, sizeof(thread_args_t));
+            args->magic = THREAD_ARGS_MAGIC;
+            if (args->magic != THREAD_ARGS_MAGIC) {
+                sprintf(log_msg, "Server: Invalid arguments structure! Magic value: 0x%X\n",args->magic);
+                log(log_msg);
+            }
+            else {
+                args->client_fd = client_fd;
+                args->protocol_type = protocol_type;
+                ret = pthread_create(&thread, NULL, handleConnections, (void *)args);
+                if (ret==0) 
+                {
+                    pthread_detach(thread);
+                }
             }
         }
     }
```
# Impact
This bug can be triggered remotely by sending crafted or rapid TCP sessions to OpenPLC on port 502 or 44818. It may cause thread crashes, segmentation faults, and denial-of-service. The issue affects default configurations and is exploitable without authentication.

thiagoralves, 2025-04-23:
Thanks for pointing this out. Could you open a PR with the fix please?

d00gman, 2025-04-23:
Yep will do. Great project and glad to help out!

thiagoralves, 2025-04-23:
Awesome! Thanks @d00gman!!